saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #88: Can't hide progressbar in no ui command line access

alexaport, 2021-12-11:
Hello,
I'm implementing this to scan in my c# application but when I start on scan  I get this transferring data window, I want to hide it and focus on my app with my internal progress bar. 

Link of current issue window: **https://prnt.sc/22r93ct**

I'll be thankful if you can help me with this.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2021-12-11:
Hello, @alexaport 
See the [Alternative User Interfaces](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2300.html) and the Saraff.Twain.Sample2 from the [Saraff.Twain.NET CS Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/cs.html)

alexaport, 2021-12-11:
Getting the same window in example 2 also.
Check the screenshot below :
https://prnt.sc/22r9m2u

I'll be thankful if you can guide me to hide this progress bar or tell me where to edit the source code of Saraff.Twain.dll

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2021-12-11:
Try set the CAP_INDICATORS capability to the FALSE (see [Strongly typed accessing to a capabilities](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2110.html) )
# CAP_INDICATORS
## Description
If TRUE, the Source displays a progress indicator during acquisition and transfer, regardless of
whether the Source's user interface is active. If FALSE, the progress indicator is suppressed if the
Source's user interface is inactive.
The Source displays device-specific instructions and error messages if either the user interface or
progress indicator is turned on. In this case it returns TWCC_OPERATIONERROR to alert the
application that it handled the error, and communicated the problem to the user.
If both the user interface and progress indicator are turned off, then the Source never displays any
message to the user, even if TWCC_OPERATIONERROR is returned. Messages to the user are under
the sole control of the Application.
## Application
If the application plans to enable the Source with TW_USERINTERFACE. ShowUI = FALSE, it can
also suppress the Source’s progress indicator by using this capability.
## Values
- **Type:** TW_BOOL
- **Allowed Values:** TRUE or FALSE

_TWAIN 2.4 Specification 10-63_

alexaport, 2021-12-11:
it worked.
Thanks is not enough but really really thanks a lot for quick and super support

closed 2021-12-11 by alexaport