saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #89: Feeder scanning wrong height document

alexaport, 2021-12-13:
Hello,
I'm trying to scan documents from the feeder but in the blank, it's giving perfect size but when I put any a4 document or any other document then the scanned image output's height is very big with blank spaces.

Please check this image to get idea: **https://we.tl/t-gjqWqH6aCm**

However on flat scan, program is working fine and scanning perfectly but I want to use the automatic feeder.

I have attached the program below. Any help will be appreciated. 

**Programm :**
        using (Twain32 _twain32 = new Twain32())
        {

            _twain32.ShowUI = false;
            _twain32.IsTwain2Enable = true;
            _twain32.OpenDSM();
            _twain32.SourceIndex = 0;
            _twain32.OpenDataSource();
            _twain32.Capabilities.XResolution.Set(200);
            _twain32.Capabilities.YResolution.Set(200);

            var _pixels = _twain32.Capabilities.PixelType.Get();

            var _val = (TwPixelType)_pixels[0];
            _twain32.Capabilities.PixelType.Set(_val);

            _twain32.EndXfer += WriteScannedImage;
            _twain32.AcquireCompleted += ScaneProcessCompleted;
            _twain32.Capabilities.Indicators.Set(false); 
            _twain32.Acquire();
     }

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2021-12-13:
Hello @alexaport 
Try set the [Image Layout](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2200.html) property and see follow capabilities:
# ICAP_SUPPORTEDSIZES
## Description
For devices that support fixed frame sizes. Defined sizes match typical page sizes. This specifies
the size(s) the Source can/should use to acquire image data.

_10-202 TWAIN 2.3 Specification_

# ICAP_AUTOSIZE
## Description
Force the output image dimensions to match either the current value of ICAP_SUPPORTEDSIZES
or any of its current allowed values.

_10-122 TWAIN 2.3 Specification_

# ICAP_UNITS
## Description
Unless a quantity is dimensionless or uses a specified unit of measure, ICAP_UNITS determines
the unit of measure for all quantities.

_10-208 TWAIN 2.3 Specification_

# ICAP_PHYSICALHEIGHT
## Description
The maximum physical height (Y-axis) the Source can acquire (measured in units of
ICAP_UNITS).
This is a read only capability.

_TWAIN 2.3 Specification 10-189_

# ICAP_PHYSICALWIDTH
## Description
The maximum physical width (X-axis) the Source can acquire (measured in units of
ICAP_UNITS).
This is a read only capability.

10-190 TWAIN 2.3 Specification

# ICAP_XSCALING
## Description
All the X-axis scaling values available. A value of ‘1.0’ is equivalent to 100% scaling. Do not use
values less than or equal to zero.

_10-212 TWAIN 2.3 Specification_

# ICAP_YSCALING
## Description
All the Y-axis scaling values available. A value of ‘1.0’ is equivalent to 100% scaling. Do not use
values less than or equal to zero.
There are no units inherent with this data as it is normalized to 1.0 being “unscaled.”

_TWAIN 2.3 Specification 10-215_

# CAP_FEEDERENABLED
## Description
If TRUE, Source must acquire data from the document feeder acquire area and other feeder
capabilities can be used. If FALSE, Source must acquire data from the non-feeder acquire area and
no other feeder capabilities can be used.

_10-56 TWAIN 2.3 Specification_

# CAP_AUTOFEED
## Description
If TRUE, the Source will automatically feed the next page from the document feeder after the
number of frames negotiated for capture from each page are acquired. CAP_FEEDERENABLED
must be TRUE to use this capability.

_10-18 TWAIN 2.3 Specification_

alexaport, 2021-12-13:
Unfortunately, none of them are working.
in image layout, I can get but can't set size and getting the exceptional error " It worked"
Also tried all other capabilities but nothing worked.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2021-12-13:
In this case try ask question to a customer service of the scanner manufacturer.

alexaport, 2021-12-13:
Sure. Thanks for your support.

closed 2021-12-13 by alexaport