saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #90: CAP_CUSTOMDSDATA set custom data

hilmoinb, 2022-02-24:
Hello,

I try so set custom data with capability [here](https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/issues/78#issuecomment-712816598)

The goal is with parameter, transfer custom data to the scanner with this parameters TwDG.Control, TwDAT.CustomDSData, TwMSG, TW_CUSTOMDSDATA).

Is it possible with the framework ?

Thanks in adavance !!

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2022-02-25:
Hello, @hilmoinb 
In the 8ce48f446c92faef8124349e5c5d8d0e95c0ae59 commit, I implemented this. 
Try to build from source and check. If successful, I will publish the nuget package. 

```C#
if(_twain32.Capabilities.CustomDSData.GetCurrent()) {
    _twain32.CustomData = new byte[] { 1, 2, 3, 4, 5, 6, 7, 8, 8 };
    var _data = _twain32.CustomData;
}

```

hilmoinb, 2022-02-25:
Ok how can i test with build please ? I usualy import the version by nuget package

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2022-02-25:
download zip-archive from github
unpack
in vs add exist project to your solution
add reference in your project (you must remove the nuget-package before)

closed 2022-03-02 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A

Nilsolen, 2022-03-02:
Sorry i'm late to awnser, i tested it yesterday night and it's ok after a quick fix : 
twain32.cs 
 TwRC _rc = this._dsmEntry.DsInvoke(this._AppId, this._srcds, TwDG.Control, TwDAT.CustomDSData, TwMSG.Get, ref _data);
replace by 
 TwRC _rc = this._dsmEntry.DsInvoke(this._AppId, this._srcds, TwDG.Control, TwDAT.CustomDSData, TwMSG.**Set**, ref _data);

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2022-03-03:
Hello, @Nilsolen 
I made some changes, see 172e8ed3f48d721309eaf89beefe1e226f0138df and 666488bf98a642c6102c8f6ebf6ea4dafb0dc741
The package has publushed, see https://www.nuget.org/packages/Saraff.Twain/1.0.31.729