saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #93: TwainException Unrecognized MSG DG DAT combination.

lemfaki, 2022-07-01:
Hello there.. I have created an app which works like a charm with your amazing library.. I tried using a second scanner for feedback but i am getting this error . 

TwainException Unrecognized MSG DG DAT combination. 
    at Saraff.Twain.Twain32._RegisterCallback()
   at Saraff.Twain.Twain32.OpenDataSource()

Scanner i am using is Epson DS-410. 
Before calling OpenDataSource() i am checking 
twain321.IsTwain2Supported
twain321.IsTwain2Enable 

both return true

Thank you in advance.. i really love your work

lemfaki, 2022-07-01:
Update!!!! 
I tried setting the ShowUI to False and now i don't get the Error message BUT. 
Trying this :  var _pixels = _twain32.Capabilities.PixelType.Get(); 

2022-07-01 15:50:29.9472|Error|Saraff.Twain.TwainException: The data source is not open. Источник данных не открыт.
   at Saraff.Twain.Twain32._GetCapCore(TwCap capability, TwMSG msg)
   at Saraff.Twain.TwainCapabilities.Capability`1.Get()

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2022-07-01:
Hello, @lemfaki 
Thanks for your feedback!
About your issue, apparently, your data source does not support TWAIN2 mode, although it reports this. Try check it out using TWAIN1.x.

lemfaki, 2022-07-01:
It seems so yes.. Do you have any TWAIN1 samples i can try out?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2022-07-01:
See [Use TWAIN 2.x](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w3000.html) wiki page, [Saraff.Twain.NET CS Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/cs.html) and [Saraff.Twain.NET Outproc Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/outproc.html)

lemfaki, 2022-07-01:
With the [Saraff.Twain.NET Outproc Samples] i can see that Twain 1.x can read the values of the scanner while 2.x indeed recieves the same error as me..

lemfaki, 2022-07-01:
Sorry for the spamm.. I found the problem.. Didn't have the latest TWAINDSM.dll files.. After i updated them, it worked like a charm

closed 2022-07-05 by lemfaki