saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #95: Scanning too many pages will result in too much memory. Cannot release during a scan.

a188658587, 2022-08-12:
扫描过多页会导致内存占用过多。在一次扫描过程中无法释放。
Scanning too many pages will result in too much memory. Cannot release during a scan.

a188658587, 2022-08-12:
[image]

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2022-08-12:
Hello, @a188658587 
See follow:

- [Using a IStreamProvider for caching data of acquired image](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2411.html)
- [Using a IImageHandler for processing acquired a image data from unmanaged memory to a stream](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2412.html)
- [Using a IImageFactory for processing acquired a image data from a stream to a image](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2413.html)
- [Buffered Memory Mode Transfer](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2430.html)
- [Disk File Mode Transfer](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2420.html)

a188658587, 2022-08-15:
IoC is difficult for me at present, but it helps me a lot. I have found a solution. Thank you very much.

closed 2022-08-15 by a188658587