saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #98: The ability to select a specific scanner is not available

lhl1314, 2022-10-24:
The default scanner has 170 support。
[image]

Choosing a specific scanner offers only 29 capabilities。
[image]

What exactly is the reason for this?
How do we solve this problem？

When I select the specific scanner cannot be generated  The picture 。

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2022-10-24:
Hello, @lhl1314 
_Sources may support a large number of capabilities but are required to support very few. To determine if a capability is supported by a Source, the application can query the Source using a DG_CONTROL / DAT_CAPABILITY / MSG_GET, MSG_GETCURRENT, or MSG_GETDEFAULT operation. The application specifies the particular capability by storing its identifier in the Cap field of the TW_CAPABILITY structure. This is the structure pointed to by the pData parameter in the DSM_Entry( ) call._

TWAIN 2.3 Specification 10-1

lhl1314, 2022-10-24:
Are you saying that I should change this part of the source code to get and set capabilities?

奔跑的小白
***@***.***

 TWAIN pecification 1.0-1.9 is the default on the scanner and windows system when I debug the program。

I don't know if this has anything to do with it?

Do you mean that I need to change the source code in the part of setting capability and acquiring capability?

At this point the source code and can not select the corresponding scanner to scan. Does the source code have other methods that can be called directly?

lhl1314, 2022-10-24:
Is it impossible to do this if the data source doesn't support it?

But I was able to select and scan with other frames. I don't think it's the data source.
如果数据源不支持的话是不是就做不到这个功能呢？

奔跑的小白
***@***.***

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2022-10-24:
No, the list of capabilities depends on a Data Source.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2022-10-24:
如果数据源不支持的话是不是就做不到这个功能呢？
No, only if your application implements it.

lhl1314, 2022-10-24:
I need to set the properties of the image according to this ability. Without this ability, the setting of the image will be abnormal.

Now that I want to continue using your framework, what do I need to do?

奔跑的小白
***@***.***

lhl1314, 2022-10-24:
Where does this ability come from? Why do you get different capabilities from connecting one scanner to two scanners?

奔跑的小白
***@***.***

lhl1314, 2022-10-24:
非常感谢您的解答，我再去思考思考，如果您有时间的话，也可以想一想这个问题呐，到底是源码的问题，数据源的问题，还是吐温规范的问题。

奔跑的小白
***@***.***

closed 2022-11-02 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A