saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #99: The example cannot be downloaded

lhl1314, 2022-10-24:
[image]

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2022-10-24:
Hello @lhl1314 
I checked it. It works.
[image]

lhl1314, 2022-10-24:
Maybe the download link is in my area  Can't access ！

Could you please download it for me and send it to me via email? I need all the demos for this part as a reference. Thanks a million!

奔跑的小白
***@***.***

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2022-10-24:
https://goo.gl/doYrzd

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2022-10-24:
[Saraff.Twain.Samples.zip](https://github.com/saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET/files/9849834/Saraff.Twain.Samples.zip)

closed 2022-11-02 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A