saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #101: Set scanner page size to "match original size"

tdhillon-552, 2023-03-24:
scanner: Canon dr-c230

Hello Saraff, 

I've been working on a project to scan and upload documents using your repo. So far it's been working quite well. Thank you for sharing!

My final issue with my project is how do I set the page size to match the original size of the page?

here's my initialization code
[```c#

                _twain.OpenDSM();
                _twain.ShowUI = false;   
                _twain.SetCap(TwCap.AutoSize, TwBP.Auto);

``` 

I'm assuming once I call the twain driver it has default settings. The last line of the code is probably the issue. when I run my app i get a "The data source is not open" error.

Any help would be appreciated!

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2023-03-24:
Hello, @tdhillon-552 

The application, Source Manager, and Source must communicate to manage the acquisition of data. It is logical that this process must occur in a particular sequence. For example, the application cannot successfully request the transfer of data from a Source before the Source Manager is loaded and prepared to communicate the request.

_TWAIN 2.5 Specification 2-11_

### The Description of the States
The following sections describe the states.

- State 1 — Pre-Session

The Source Manager resides in State 1 before the application establishes a session with it.
At this point, the Source Manager code has been installed on the disk but typically is not 
loaded into memory yet.
The only case where the Source Manager could already be loaded and running is under 
Windows because the implementation is a DLL (hence, the same instance of the Source 
Manager can be shared by multiple applications). If that situation exists, the Source Manager 
will be in State 2 or 3 with the application that loaded it.

- State 2 — Source Manager Loaded

The Source Manager now is loaded into memory. It is not open yet.
At this time, the Source Manager is prepared to accept other operation triplets from the 
application. 

- State 3 — Source Manager Open

The Source Manager is open and ready to manage Sources.
The Source Manager is now prepared to provide lists of Sources, to open Sources, and to close 
Sources.
The Source Manager will remain in State 3 for the remainder of the session until it is closed. 
The Source Manager refuses to be closed while the application has any Sources open.

- State 4 — Source Open

The Source has been loaded and opened by the Source Manager in response to an operation 
from the application. It is ready to receive operations.
The Source should have verified that sufficient resources (i.e. memory, device is available, 
etc.) exist for it to run.
**The application can inquire about the Source’s capabilities** (i.e. levels of resolution, support of 
color or black and white images, automatic document feeder available, etc.). The application 
can also set those capabilities to its desired settings. For example, it may restrict a Source 
capable of providing color images to transferring black and white only. 
Note: Inquiry about a capability can occur while the Source is in States 4, 5, 6, or 7. But, an 
application can set a capability only in State 4 unless special permission is negotiated 
between the application and Source.

- State 5 — Source Enabled

The Source has been enabled by an operation from the application via the Source Manager 
and is ready for user-enabled transfers.
If the application has allowed the Source to display its user interface, the Source will do that 
when it enters State 5.

- State 6 — Transfer is Ready

The Source is ready to transfer one or more data items (images) to the application. 
The transition from State 5 to 6 is triggered by the Source notifying the application that the 
transfer is ready.
Before initiating the transfer, the application must inquire information about the image 
(resolution, image size, etc.). If the Source supports audio, then before transferring the image, 
the Application must transfer all the audio snippets that are associated with the image.
It is possible for more than one image to be transferred in succession. This topic is covered 
thoroughly in Chapter 4, "Advanced Application Implementation".

- State 7 — Transferring

The Source is transferring the image to the application.
The transfer mechanism being used was negotiated during State 4. 
The transfer will either complete successfully or terminate prematurely. The Source sends the 
appropriate Return Code indicating the outcome.
Once the Source indicates that the transfer is complete, the application must acknowledge the 
end of the transfer.

_TWAIN 2.5 Specification 2-13_

You must invoke the Twain32.OpenDataSource() before setting the capability.
Therefore, you must invoke follow method sequence
```C#
_twain32.OpenDSM()
// ...
_twain32.OpenDataSource()
// ...
_twain32.SetCap(TwCap.AutoSize, TwBP.Auto);
// ...
```

Also, you can see 
- [Alternative User Interfaces](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2300.html)
- [Strongly typed accessing to a capabilities](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2110.html)
- [General (not strongly typed) accessing to a capabilities](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2120.html)

tdhillon-552, 2023-03-24:
ah I see. I appreciate the detailed response. Thank you!

closed 2023-03-24 by tdhillon-552