saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #104: After data source is chose, no UI is show

pavleda, 2023-07-07:
Hello,

I use TwainX with DS from your repo. After I chose ShowUI and chose Saraff DS, the UI just blinks and disappears. 

Do you have an idea what is the cause of the problem?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2023-07-08:
Hello, @pavleda 
What kind of DS are you talking about? How did you install it? Maybe you are using multithreaded access?

pavleda, 2023-07-10:
@saraff-9EB1047A4BEB4cef8506B29BA325BD5A 

I installed DS from your repo via MSI. I try it on your sample, where you use single thread access

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2023-07-11:
Hello, @pavleda 
I has downloaded the [Saraff.TwainX.NET Core Sample2](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/download/Saraff.TwainX.NETCoreSample2.zip) (NEW. 28.01.2018) and built it (in VS2022). It work fine. Screenshort below.

[image]

I updated nuget packages to latest version and built it (in VS2022). It work fine. Screenshort below.

[image]

I has downloaded the [Saraff.TwainX.NET Core Sample1](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/download/Saraff.TwainX.NETCoreSample1.zip) (NEW. 28.01.2018) and built it (in VS2022). It work fine. Screenshort below.

[image]
[image]
[image]

I updated nuget packages to latest version and built it (in VS2022). It work fine. Screenshort below.

[image]
[image]

pavleda, 2023-07-12:
I also done that, and I got this: https://imgur.com/a/IJQs8mB

I'm not sure what is problem exactly

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2023-07-12:
The DS requires a message loop to display a window, which is missing in a console application. Therefore the scan does not start. This is a peculiarity of this DS.

pavleda, 2023-07-12:
Oh, I see. So what sample to use to display DS UI screen?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2023-07-12:
The [Saraff.TwainX.NET WinForms Sample1](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/download/Saraff.TwainX.WinFormsSample1.zip)

closed 2023-07-14 by pavleda