saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #111: TWAIN 2.1 driver not works with Stwain.IsCapSupported(TwCap.FeederLoaded

Lee-Shadmi, 2024-04-03:
I am using this code to detect the card on the feeder:
Stwain.IsCapSupported(TwCap.FeederLoaded).
It is working fine with some devices but in other not, i understand that the device that it is not working with using a twain driver of TWAIN 2.1
Is there any info about using 2.1 version that is not the same at the older twain versions?
do i missing a step in my code?

Again - all works fine with other devices but not with this one.

Lee-Shadmi, 2024-04-03:
Even if i will scan the _twain_AcquireCompleted event will not triggered.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2024-04-04:
Hello, @Lee-Shadmi 
Try update TWAIN DSM up to 2.3 (or 2.4) version. See [Installation of the Data Source Manager](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w0200.html)

closed 2024-04-09 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A