saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #112: Disk File Mode doesnt fill the File

31Kapitano, 2024-05-24:
Hello, 

I'm new to this area.

I use the "Disk File Mode Transfer" and I followed the instructions in the description. 

After the "_twain32_SetupFileXferEvent" is executed, a file is created and the scanner begins scanning. But the file remains empty and I can't find the error.

My Code: 

_twain.OpenDSM();
_twain.OpenDataSource();

_twain.Capabilities.XferMech.Set(TwSX.File);

if (_twain.Capabilities.XferMech.GetCurrent() is TwSX.File)
{
    _twain.SetupFileXferEvent += _twain32_SetupFileXferEvent;
    _twain.FileXferEvent += _twain32_FileXferEvent;
}

_twain.ShowUI = true;
_twain.Acquire();

private void _twain32_SetupFileXferEvent(object sender, Twain32.SetupFileXferEventArgs e)
{
    try
    {
        string sDate = DateTime.Now.ToString("ddMMyyyyhhmmss");
        iCounter++;
        _sFile = @"C:\\TwainTest\" + sDate + "_" + iCounter + ".";

        e.FileName = _sFile + this._twain.Capabilities.ImageFileFormat.GetCurrent().ToString().ToLower();
       
    }
    catch (Exception ex)
    {
        //MessageBox.Show(ex.Message, ex.GetType().Name, MessageBoxButtons.OK, MessageBoxIcon.Error);
    }
}

private void _twain32_FileXferEvent(object sender, Twain32.FileXferEventArgs e)
{
    try
    {        
        var _img = new BitmapImage(new Uri(e.ImageFileXfer.FileName));

        //var img = Image.FromFile(e.ImageFileXfer.FileName);
                        
        
        if (this.CurrentBitmap != null)
        {
            this.CurrentBitmap.Save(_sFile + "jpg",ImageFormat.Jpeg);
        }
        
    }
    catch (Exception ex)
    {
        
    }
}

Where is my Mistake?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2024-05-25:
Hello, @31Kapitano 
See [Disk File Mode Transfer](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2420.html) wiki page.
Also you can see the Saraff.Twain.Sample2 of [Saraff.Twain.NET CS Samples](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/samples/cs.html)

31Kapitano, 2024-05-25:
I did it. I've looked at the wiki and the examples. The code I posted above is based on that. I've also looked at the Twain documentation. But I still can't find the error.

As already mentioned, the EventHandler creates an empty file and the scanner starts scanning. But the file remains empty.

Probably just another command or call is missing that writes the data from the scanner into the file.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2024-05-25:
The data is writing in the file by the DS (scanner).
So, if the Saraff.Twain.Sample2 just works that try checking permissions on target directory.

About follow frame:
``` C#
if (_twain.Capabilities.XferMech.GetCurrent() is TwSX.File)
{
_twain.SetupFileXferEvent += _twain32_SetupFileXferEvent;
_twain.FileXferEvent += _twain32_FileXferEvent;
}
```

There is no need to do checking XferMech, you just need to subscribe to events

31Kapitano, 2024-05-27:
I think you are right. It will definitely depend on the permissions. 

Thanks for your support.

closed 2024-05-27 by 31Kapitano