saraff-9EB1047A4BEB4cef8506B29BA325BD5A/Saraff.Twain.NET #113: Is Aquire an async operation

gourango-modak, 2024-06-11:
I have a WinForms application where I scan images one by one and display all images in a form using a loop until the user stops the loop. However, when I invoke the Acquire method, it opens the device-specific settings UI, but before the scan completes, it goes to the next line and continues the loop. Therefore, I am unable to acquire any images. How do I call the Acquire method synchronously, so that after one scan completes, it opens the UI again and scans the next image? Please help me how can I achieve this functionality.

Application: WinForms application using Visual Studio
Library: Saraff.Twain.NET v1.0.32.732

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2024-06-14:
Hello, @gourango-modak 
Regarding the first issue, you need to set the Twain32.DisableAfterAcquire property to false and handle the Twain32.EndXfer event (see [Native Mode Transfer](https://saraff-9eb1047a4beb4cef8506b29ba325bd5a.github.io/sarafftwain/wiki/w2410.html) ) instead of calling the Twain32.Acquiry method in a loop.

Regarding the second issue, the scanner window may be trying to adjust the screen settings to ensure proper operation. Try using a different scanner driver or set the default screen settings (96 dpi or 100% scaling)

closed 2024-06-19 by saraff-9EB1047A4BEB4cef8506B29BA325BD5A

gourango-modak, 2024-06-20:
@saraff-9EB1047A4BEB4cef8506B29BA325BD5A 

Thank you so much for your guidance. I am facing another issue when trying to scan multiple times using Twain32.DisableAfterAcquire = false. I get the following exception:

`Attempted to read or write protected memory. This is often an indication that other memory is corrupt.`

Could you please explain what might be causing this issue?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2024-06-20:
Hello, @gourango-modak 
The problem is probably in the use of asynchronous calls. Access to GUI elements should only be possible from the thread that created it.
Try handle Twain32.AcquireCompleted event to signal that the operation has finished.

gourango-modak, 2024-06-20:
@saraff-9EB1047A4BEB4cef8506B29BA325BD5A 

Thanks for the prompt reply. I tried your suggested solution, but I am still encountering the same exception. It appears that the issue is related to the use of asynchronous calls.

Could you please provide guidance on how to meet the following requirements -
In my Windows Forms Application, after each image scan completes, I need to display the scanned image in a form with multiple sections, where each section displays a scanned image. Additionally, there are other actions that need to be performed after each image scan completion.

[image]

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2024-06-20:
the Twain32.EndXfer event invoke for every transfer to indicate has received all the data it expected.
```C#
private void _twain32_EndXfer(object sender,Twain32.EndXferEventArgs e) {
    try {
        using(var _image = e.Image) {
            _image.Save("filename.tif",ImageFormat.Tiff);
        }
   } catch(Exception ex) {
        MessageBox.Show(ex.Message,ex.GetType().Name,MessageBoxButtons.OK,MessageBoxIcon.Error);
    }
}
```
the Twain32.AcquireCompleted event is called after all images have been transferred.
```C#
private void _twain32_AcquireCompleted(object sender,EventArgs e) {
    try {
        for(int i=0; i<this._twain32.ImageCount; i++) {
            using(var _image=this._twain32.GetImage(i)){
                _image.Save(Path.GetTempFileName());
            }
        }
    } catch(Exception ex) {
        MessageBox.Show(ex.Message,ex.GetType().Name,MessageBoxButtons.OK,MessageBoxIcon.Error);
    }
}
```

gourango-modak, 2024-06-26:
@saraff-9EB1047A4BEB4cef8506B29BA325BD5A 

This solution only works when the `_twain32_EndXfer` callback method is called, but I want to be able to cancel it from outside of this method.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2024-06-26:
@gourango-modak 
> This solution only works when the `_twain32_EndXfer` callback method is called, but I want to be able to cancel it from outside of this method.

This is impossible.

gourango-modak, 2024-06-28:
Hello @saraff-9EB1047A4BEB4cef8506B29BA325BD5A 

I am using the following virtual scanner to test this library. However, when I click the cancel button in the scanner UI, I don't receive a cancel exception from the `Acquire` method. However, I get an exception from the [Dynamsoft library](https://www.dynamsoft.com/dotnet-twain/features/). What might be causing this issue?

On the other hand, with the following TWAIN import driver, I receive the cancel exception from the `Acquire` method of Saraff.Twain.

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2024-06-29:
Hello, @gourango-modak 
The Twain32.Acquire method should not block the form, because it must be asynchronous (the exception is console applications, since they do not have an message loop). You can use the Twain32.Parent property to block user input into parent form.

To handle exceptions from the DataSource (thrown during scanning), you must use the Twain32.AcquireError event

gourango-modak, 2024-07-02:
Hello, @saraff-9EB1047A4BEB4cef8506B29BA325BD5A 

For this specific device [TWAIN import driver (XPCTWAIN TIFF Multipage/Multifile TWAIN Import Driver Demo Version)](https://www.jse.de/download.html), it blocks UI interaction with the form behind it. Please refer to the video below for a demonstration.

https://drive.google.com/file/d/1yURWaPc72MCizcZuTeJ-689n0kOgNt5T/view

We've set the Twain32.Parent property to keep the device UI always on top of the parent form. This behavior is specific to this device. Is there a way to modify the library so that it behaves like other devices?

Another issue observed with this device is that when we call the Acquire method, it becomes a synchronous operation until the user interacts with the device UI. Additionally, the Twain state doesn't transition to 'DSEnabled' immediately upon opening; it transitions to 'DSEnabled' only when the 'Scan' button is clicked. Furthermore, the data source state becomes disabled immediately after the scan completes, even though the device UI remains open.

Is there a possibility to address these issues by modifying the library?

saraff-9EB1047A4BEB4cef8506B29BA325BD5A, 2024-07-03:
Hello @gourango-modak 
> For this specific device [TWAIN import driver (XPCTWAIN TIFF Multipage/Multifile TWAIN Import Driver Demo Version)](https://www.jse.de/download.html), it blocks UI interaction with the form behind [...]

This is completely normal behavior. That's how it should be.

> We've set the Twain32.Parent property to keep the device UI always on top of the parent form. This behavior is specific to this device. Is there a way to modify the library so that it behaves like other devices?

Both the Saraff.Twain32.NET library and the DataSource behave as they should in this situation. There's no reason to change anything.

> Another issue observed with this device is that when we call the Acquire method, it becomes a synchronous operation until the user interacts with the device UI. 

It depends on the DataSource. The Saraff.Twain.NET library only interacts with the device in accordance with the TWAIN specification.

> Additionally, the Twain state doesn't transition to 'DSEnabled' immediately upon opening; it transitions to 'DSEnabled' only when the 'Scan' button is clicked. Furthermore, the data source state becomes disabled immediately after the scan completes, even though the device UI remains open.

That's how it should be. Otherwise, it would be impossible to get or set the DataSource Capabilities.

> Is there a possibility to address these issues by modifying the library?

There is no reason for this. Everything works like clockwork.
See [TWAIN Specification](https://twain.org/specification/)